LumaTeam/Luma3DS #938: System crashes when clicking SD on FBI

Gupple, 2017-11-18:
**System model:**

[Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[11.6.0-39U SysNAND]

**Entrypoint:**

[Boot9Strap]

**Luma3DS version:**

[8.1.1 stable]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: ( )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: (x)

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
System shows an ARM9 exception when clicking SD on the FBI menu. Updated luma to 8.1.1 and system crashes, showing an ARM9 data abort.

Tyulis, 2017-11-18:
"Data abort" shows a problem in the program's data, so it is more likely a problem in FBI rather than in Luma

Gupple, 2017-11-18:
So should I try to reinstall FBI?

urherenow, 2017-11-19:
Couldn't hurt. iirc FBI had updates related to b9s, and you didn't mention which version of FBI was installed.

Gupple, 2017-11-19:
It was the latest version of FBI.

urherenow, 2017-11-19:
So now it's time to run tests on the SD card. I have no such issue on O3DS or N3DS. If it passes all tests, you might try formatting the card and then copy everything back to it.

Hikari-chin, 2017-11-20:
Can't reproduce. FBI works fine for me as both a CIA and 3DSX.

closed 2017-11-20 by Hikari-chin