LumaTeam/Luma3DS #941: ARM11 Crash with NTR Selector's mode 3

Nikslg, 2017-11-21:
**System model: **
old 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.6.0-39E

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
8.1.1. Stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

After updating Luma to 8.1.1 and NTR Boot Selector mode 3 to 2.9, I was unable to start MH XX translation patch with Mode 3 NTR, because it would crash whole console.

**Steps to reproduce:**

1. Start Mode 3 NTR Selector

2. Start MH XX

3. Get Crash

**Dump file:**
[crash_dump_00000007.zip](https://github.com/AuroraWright/Luma3DS/files/1492997/crash_dump_00000007.zip)

Hikari-chin, 2017-11-22:
Did you update to the latest commit or 8.1.1 stable? NTR is incompatible with the sysmodule replacements in the last few commits and will crash/error on startup until it is updated by 44670. 

Additionally, did you try using the older version of BootNTR Selector that you were using before you updated it and Luma?

Nanquitas, 2017-11-22:
NTR 3.2 and higher works fine with last commits.
The issue is most likely from your plugin you should contact its author.

Also, note that sometimes the plugin injection can fail and simply rebooting solves it.

Anyway, it's not related to Luma3DS itself.

Nikslg, 2017-11-22:
Hikari-kun, tried in both. In most recent as you said it would crash on start up of NTR. But game launch crash was on stable version.
Nanquitas, The thing is, it was working fine before updating. But it's not really a plugin, it's a game patch that can be started even with Luma Game Patching. However with NTR selector mode 3 it would crash on start up of Monster Hunter XX.

Nanquitas, 2017-11-22:
Hmm, do you mind listing and linking what you use (patch & plugin) ?

Hikari-chin, 2017-11-22:
Whoops, misread.

Nikslg, 2017-11-22:
https://gbatemp.net/threads/release-monster-hunter-double-cross-english-ui-patch.465296/
This Is the only thing I am using.
Usually I would just start NTR mode 3, then start MHXX. Before that everything was working fine.
I assume it could be memory issue as well, since game itself is quite big. I mean, this game restarts console when you quit the game on home menu

Nanquitas, 2017-11-22:
Try to use the luma's patching method available on the thread you linked instead of the plugin maybe.

Nikslg, 2017-11-22:
Yes, I use Luma Game Patching now, but still unable to launch NTR mode 3 with MHXX. I use it for screenshot and such.

Shadow322, 2017-11-22:
Luma3DS has built in screenshot ability and is way superior to NTR

Nikslg, 2017-11-22:
How to use it? Now I am curious :D

Shadow322, 2017-11-22:
Open Rosalina menu

TheProgramerEX, 2017-11-28:
I know no one has been on this for almost a week, but I wanted to ask a question to Nikslg 3 things:

1. Have you tried using any other patches for games and/or plugins to see it works?

 2. Have you tried downgrading to your previous version of ntr to check if it works?

3. If so and it still crashes, can you downgrade your luma version and get your ntr version to 2.9 again manually to make sure that ntr version was working on previous versions of luma?

ericjwg, 2017-12-01:
may not be the same issue...
arm11 crash with Pokemon cro file.

**System model: **
new 3DS XL

SysNAND version (+emu/redNAND version if applicable):
11.6.0-39J, U

Entrypoint (How/what you're using to boot Luma3DS):
Boot9Strap

Luma3DS version:
8.1.1 release and newest commit

Luma3DS configuration/options:

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: ()

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

Explanation of the issue:
game crashes right after launched, no title animation.

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1522313/crash_dump_00000000.zip)

closed 2017-12-10 by Hikari-chin