LumaTeam/Luma3DS #947: Small feature/polish request: PIN timer

LinkSoraZelda, 2017-11-30:
The console will not turn off the screen while the lid is closed during usage of arm9 payloads, so, (especially new 3DS, since objects can press against the outward facing power button in your pocket/bag) booting will keep the screens on and drain the battery without the user aware. 

If a PIN is registered on the console, it will wait for an indefinite amount of time before continuing. A simple timer which will shut the system down after a certain amount of time a PIN has not been entered will be useful to prevent dead batteries.

Kaisogen, 2017-12-05:
Yeah, this issue is annoying on N3DS. I have my 3DS turn on in my bag all the time, and I end up having a drained battery before I get home, which drives me nuts. 

A timer should have a consistent limit of 2.5 minutes before shutdown. Anything more is too long, as the user is given enough time to read the message before the 3DS turns off.

Hikari-chin, 2017-12-18:
Because of maintenance costs and potential user inconvenience, this will not be implemented. However, we are considering implementing closed shell detection on the config and chainloader menus to same effect.

closed 2017-12-18 by Hikari-chin

Hikari-chin, 2017-12-18:
Shell detection has been implemented as of 6a0f332.