LumaTeam/Luma3DS #948: Problems in ALL games w/ 3DS XL

DenkDoritos, 2017-12-01:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: (X )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
I'm having FUCK TON problems with my old N3DS XL i bought on 2011, on Europe.
Since this year's Summer, i've hacked it after my other N3DS XL that doesn't show me errors regularly. Well, what's the problem? The problem's day zero was two hours after installing B9S on Luma 8.1, with Luigi's Mansion. I'd think it was an .CIA error, so i deleted that and installed other .CIA.
Then, like 3 months later i noticed that an error like the same occured on 1/6 out of the gravity, on Project Mirai DX. Later, on other songs; then to Mario Kart 7, and finally with all the games on the SD. The SD was bought from a chinese website, a Kingston 16GB microSDHC. I swap adapters, checked if Enable Game Patching did something, reinstalled Luma's payload (boot.firm.)
I'd appreciate if you could tell me what's happening, i will leave you all the steps and etc. Everything but the err code; i'll insert the error on Mirai DX's 1/6 out of gravity.

**Steps to reproduce:**

1.For Mario Kart: Open it
    For Mirai DX: Open the app and go to musical menu
    For AC: New Leaf (WA): Give a gift to Morania
    For Hakuoki: No errors shown till now.
    For Rythm Thief: Open it.
    For OoT: Open it
    For Majora: Open it and let it create save data.

**Dump file:**

           Sorry, no access to a computer ftw.

DenkDoritos, 2017-12-01:
Btw, forgot brightness on 1, everything is blank since that

DenkDoritos, 2017-12-01:
And Sys 11.6.0-39E

urherenow, 2017-12-02:
Your sd card is faulty or fake.

Derek1199, 2017-12-02:
^ @urherenow is right. That is your problem @DenkDoritos .

closed 2017-12-03 by Hikari-chin