LumaTeam/Luma3DS #949: Luma cannot patch romfs in applets

ctrninja, 2017-12-02:
(Hello there, I'm the creator of Foxverse) While patching Miiverse, I noticed that Luma cannot patch romfs inside applets this is a big issue for us since in order to get HTTPS patched inside Miiverse we used LayeredFS for this, since this doesn't seem to work we have to use 2 CIAs to patch this and 3DS system updates no longer work because of this, I consider this being a enhancement to LayeredFS. Thank you. -ninjafox

ihaveamac, 2017-12-02:
as far as I know this won't happen due to code differences in applets, compared to normal applications.

urherenow, 2017-12-02:
What exactly is miiverse being patched to do? Use a different server? Because... isn't miiverse supposed to be discontinued?

HoLLy-HaCKeR, 2017-12-02:
@urherenow yes, it's common for custom servers to patch https URLs in the client so the requests can be redirected to a new one without having to fuck with host redirection and installing custom certificates.

profi200, 2017-12-02:
You will need custom certs anyway if you want to keep security. Patching https completely out is just dumb.

ctrninja, 2017-12-02:
@urherenow Yes, It's getting patched to use a different server.

ctrninja, 2017-12-02:
@profi200 Originally we wanted to use HTTPS but it didn't work because of this issue we had.

khang06, 2017-12-02:
looks like trying to use game patching on home menu is also affected by this

ctrninja, 2017-12-04:
Updating the title as it seems that not even system data archives work with LayeredFS.

frozenLake, 2017-12-05:
Oh, this is why my RootCA edits were not doing anything? Okay, this is something I have a decent need for.

frozenLake, 2017-12-30:
You know, I'm curious: Has any research been done regarding this? Is there anything of note that is making this more difficult on the devs?

Brawl345, 2019-06-30:
Any news for this? Would like to add Let's Encrypt to the 3DS browser by editing the rootca.pem, but LayeredFS doesn't work with applets.

ParzivalWolfram, 2019-08-23:
Would be handy for patching several features, more support and new certs into the Browser.

billyohgren, 2020-09-08:
I would love to patch the home menu, and if this is affected by this, could someone give me pointers on where this (layeredFS) is being used in the source code?

AuroraWright, 2020-11-17:
As @ihaveamac said sadly the SDK code patched by layeredfs isn't there on these titles, so the current approach won't work

closed 2020-11-17 by AuroraWright