LumaTeam/Luma3DS #950: mask of majora 3ds An  exception occurred

mariyo8, 2017-12-02:
I have a 3ds that has the freeshop installed. All of the games seem to work fine but with mask of majora 3ds I get this error. 

[image]

I'm new to Luma3ds so I'm not quote sure how to solve. I was given the 3ds by a friend. Any guidance would be appreciated. 

These are the files in my sd card. I'm not too sure what i'm running or how to find out...

[image]

I have the Nintendo 3ds XL

I'm running Sys 11.5.0-38u

Freeshop 2.2.0

[image]

**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

Sonlen1414, 2017-12-02:
Buy the game, it works 100% of the time.

Navioli, 2017-12-02:
1: Enable Enable loading external FIRMs and modules: ( )
2: mask of majora 10/10

mariyo8, 2017-12-02:
@Navioli 

How do I do this?

Navioli, 2017-12-02:
1. Make sure the console is off.
2. Hold select while turning on the console.
3. You should be on the Luma3DS configuration screen.
4. Go down to Show NAND or user string in System Settings and select it.
5. Go to Enable loading external FIRMs and modules and select it.
6. Go ahead and press start to leave. It should work, its what worked for me.

mariyo8, 2017-12-02:
@Navioli 
Did what you said. The screen error pops up...

Sonlen1414, 2017-12-02:
Of course the error still pops up, what he suggested you do does literally nothing for corrupted titles, Luma does not support Piracy and you won't be getting help with it here. Buy the game as I said, it is guarenteed to work. ¯\\\_(ツ)_/¯

Navioli, 2017-12-02:
Oh, he pirated his games? It started doing that when I hacked eShop titles with NTR.

closed 2017-12-03 by Hikari-chin