LumaTeam/Luma3DS #951: Rosalina: changing menu combo only recognizes one key

Tilka, 2017-12-02:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v8.1.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: (off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

When changing the key combination to open Rosalina, it's impossible to use a combination of two or more keys. Only a single key is recognized.

**Steps to reproduce:**

1. Open Rosalina.
2. Try to "change" the menu combo to the default L+Down+Select.

Hikari-chin, 2017-12-03:
Can't reproduce. Let me set my menu combo to however many keys I wanted.

closed 2017-12-03 by Hikari-chin

Tilka, 2017-12-03:
Just in case anyone else has this issue: it's important to release all buttons at the same time.

Hikari-chin, 2017-12-03:
That's not necessary on recent versions, Rosalina waits to determine the combination until you release **all** keys.

gingerbeardman, 2018-04-11:
I'm on Luma 9.0 and I still have to release all button at once @Hikari-chin