LumaTeam/Luma3DS #952: ARM 11 Crash in microSD Management when pressing power button

Natsuru3, 2017-12-03:
**System model:**
New 3DS

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
SysNAND 11.6.0-39U
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap 1.3
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.1.1 stable
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (*)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
ARM 11 Crash

**Steps to reproduce:**

1. If the power button is pressed when in microSD Management regardless if a PC was connected the system crashes. Have not tried pressing the power button during the Management setup.
(connected to network)

**Dump file:**

[arm11.zip](https://github.com/AuroraWright/Luma3DS/files/1524536/arm11.zip)

SmackBro, 2017-12-03:
Can confirm this,just tested it right now and got an arm11 crash.

HoLLy-HaCKeR, 2017-12-03:
Probably a silent crash that Nintendo ignored because you're shutting down the software anyway. The settings app had something like this too.

Hikari-chin, 2017-12-10:
Known issue, Nintendo error, not Luma.

closed 2017-12-10 by Hikari-chin