LumaTeam/Luma3DS #953: Feature Request: Default Chainloader Boot / key-pad SysNAND boot option.

Spork-Schivago, 2017-12-06:
**System model:**

[Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[10.3.0-28U SysNAND, GW3D 10.2.0-28U EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[8.1.1-f6483ec]

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (yes)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
This isn't a bug, I'm trying to submit a feature request but I'm having trouble figuring out how to add the label.   I've filled out the requested information to the best of my ability.

I have a Gateway 3DS and currently am using the Chainloader feature of Luma3DS.   I have it configured where if I hold down the "down" key on the d-pad during bootup of the 3DS, it'll boot into the Gateway mode and show the dragon, along with the EmuNAND that I have installed.

If I don't hit the "down" key on the d-pad during bootup, it boots to normal SysNAND.    I'd like to submit a feature request where I can switch it around.    For example, I would like to have it only boot to the SysNAND if I hold down the "down" button on the d-pad, otherwise, have it boot directly into the Gateway mode (or whatever file we're chainloading).

Thanks!

**Steps to reproduce:**

1.

2.

**Dump file:**

Kaisogen, 2017-12-06:
What do you mean by Gateway 3DS? Is it a 3DS previously hacked with Gateway, or was a 3DS released by Gateway?

ihaveamac, 2017-12-06:
"Gateway 3DS" usually refers to the red card, so...

Margen67, 2017-12-07:
1. Stop using Gateway
2. ???
3. Profit

mzino, 2017-12-07:
Since the global consensus in the community seems to be that GW is "objectively worse" (even if it does different things so it's not really comparable...) you probably won't get an option to override the default no-button action for the chainloader.

You should use BootCTR9 instead. The latest commits have .firm compatibility and there's a GW .firm around, so you can set a default option to load GW and another button for Luma or other homebrews.

urherenow, 2017-12-07:
Yay. So for that, I'll need to rebuild bootstrap9 again so bootctr9 will be the default, but named ctr9.firm or something, to ensure I don't break Luma by not having it as sd:/boot.firm (or does luma autodetect its own name for reboot cases (exit system settings, or MH on O3DS)?

mzino, 2017-12-07:
You could just press the Luma button each time it reboots, without rebuilding anything... Not 100% ideal but not totally absurd either.

Hikari-chin, 2017-12-10:
There are no plans for this at this time.

closed 2017-12-10 by Hikari-chin