LumaTeam/Luma3DS #955: Luma3DSv8.1.1.7z Error Message

jackthepanda, 2017-12-08:
I get an error message whenever I try to open Luma3DSv8.1.1.7z. 

"The structure of this archive is damaged."

ihaveamac, 2017-12-08:
try downloading it again.

SHA256: `6d51d86becac5ba1f640a7fc689c77e05c2e517c79d5682e5cccd38ef08214f9`
SHA1: `72302827e7876ccbcdb70451c162fbbecb87500f`
MD5: `4b8ebf266852684bd50b25ee629d7703`

urherenow, 2017-12-08:
Or install the latest 7zip, if that doesn't work.

jackthepanda, 2017-12-08:
I've tried three different unarchivers and they're all saying that there is something wrong with it. I'm going to try to download another unarchiver to see if that works.

ihaveamac, 2017-12-08:
are you using windows? latest 7-zip should open it fine. also check one of the hashes of the file.

jackthepanda, 2017-12-08:
I'm on a Mac. My windows computer is busted and I'm using an old mac with 10.8 on it. Archive Utility, Stuffit Expander and Winzip for mac didn't work, but something called Keka which is an unofficial 7-zip port worked. I looked on the 7-zip website, but it looks like they don't have an official Mac version. Doesn't matter much anyway, I should be getting my Vaio back from being repaired soon. Thanks for the help though! I really appreciate it. Happy Holidays. :D

ihaveamac, 2017-12-08:
use the unarchiver: https://itunes.apple.com/us/app/the-unarchiver/id425424353?mt=12 (non-mas download: https://dl.devmate.com/cx.c3.theunarchiver/TheUnarchiver.zip)

or use 7z from terminal (install p7zip with brew).

closed 2017-12-10 by Hikari-chin