LumaTeam/Luma3DS #957: mario&sonic olympic games rio 2016 crash

Ditello, 2017-12-08:
**System model:**
New 3ds xl
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable: 11.6.0.39E both

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

8.1.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: ( off )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: (x)

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.Launch Mario & Sonic at the olympic games rio 2016 eur

2. At 3ds screen. It crashes

**Dump file:**

An exeption occurred

Processor: Arm11
E type: Data abort
Process: Sergio (000400000017E300)

R0 E1A00000
R1 E28FC001
SP 10000000
LR 00100004
PC 00100034
CPSR 80000010

Stack dump: Nothing

Hikari-chin, 2017-12-10:
You probably need to enable locale emulation, running a EUR game on a U/J NAND.

closed 2017-12-10 by Hikari-chin

Ditello, 2017-12-10:
It is european sys and emu.
Game is european. And it occures every time.