LumaTeam/Luma3DS #958: Arm11 Crashed while I was playing Fire Emblem Fates: Conquest. I was playing in My Castle.

NaturalHarmonia, 2017-12-08:
Here's the dump.

[crash_dump_00000008.zip](https://github.com/AuroraWright/Luma3DS/files/1543717/crash_dump_00000008.zip)

I've had this happen multiple times a few days ago as well and thought it had stopped.

Hikari-chin, 2017-12-10:
Can't reproduce, but all the people that have complained of Fates crashes in the past have had damaged save data from improper editing.

closed 2017-12-10 by Hikari-chin

NaturalHarmonia, 2017-12-14:
I also just had a crash with Pokemon Ultra Moon. Will upload dump later.

Sonlen1414, 2017-12-14:
The issue is most likely you didnt remove the no outlines patch as it isn't compatible with the game update yet. Not a Luma issue.

Hikari-chin, 2017-12-14:
Played both of these games through to completion on the latest few commits without issue; the issue is likely that or improper save editing. Not Luma's problem, as Sonlen said.