LumaTeam/Luma3DS #961: Request: Streetpass Extension

piepie62, 2017-12-13:
Right now, the Streetpass data is limited to twelve games. Is it even feasible to extend this? If so, then can you that or is it out of the scope of this project? I understand if the answer to either of these is no, given that I have not seen anything about this anywhere. Thanks for any answers, though

TuxSH, 2017-12-15:
I don't know how/where/how many times the limit is implemented (knowing would take time) and I believe than even if I knew it wouldn't be feasible _a priory_.

Interesting proposition but I don't think it's feasable reasonably.

closed 2017-12-15 by TuxSH

piepie62, 2017-12-16:
Alright, that's fair. Just was wondering if you had any clue. Thanks for the reply!