LumaTeam/Luma3DS #973: Theatrhythm Final Fantasy: Curtain Call Crashing on Quest MedleySave to SD

Quantum-Quartz, 2017-12-17:
**System model:**
Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.6.0-39U
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.1.1-f6483ec
**Luma3DS configuration/options:**

Default EmuNAND: ( 1)

Screen brightness: ( 4)

Splash: ( Off)

PIN lock: ( Off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( OFF)

Use EmuNAND FIRM if booting with R: ( OFF)

Enable loading external FIRMs and modules: (OFF )

Enable game patching: ( OFF)

Show NAND or user string in System Settings: ( ON)

Show GBA boot screen in patched AGB_FIRM: ( OFF)

Patch ARM9 access: ( OFF)

Set developer UNITINFO: (OFF )

Disable ARM11 exception handlers: ( OFF)

--

**Explanation of the issue:**
In game: Theatrhythm Final Fantasy: Curtain Call
Game Mode: Quest Medley
Game crash after trying to save here.

Things I've tried that haven't worked:

- Re-installing the CIA
- Playing from the Sky 3DS again
- Creating A new Extra data
- Playing with and without the DLC
- Updating from A9LH to BS9
- Using the Nintnedo E-Shop Repair Feature.
- TikSweep to remove the existing tickets for the game in case they are bugged
- Creating a new save data in case my save data is bugged.
I
've also tried certain combinations of all of the above fixes but maybe I'm missing something. 

EDIT: A picture of the error: https://imgur.com/a/2ItsB

**Steps to reproduce:**

1. Install Theatrhythm Final Fantasy: Curtain Call CIA via Nintendo E-Shop or via CIA

2. Play until you unlock Quest Medley

3.  play 1-3 songs in Quest Medley

4. Game will Crash during the saving screen with a chocobo running in the bottom right.

**Dump file:**

[image]

Sonlen1414, 2017-12-17:
Could you upload your save file?

Quantum-Quartz, 2017-12-18:
Attached are my exrtra data and the save file
[TFFC_EXTRA.zip](https://github.com/AuroraWright/Luma3DS/files/1566245/TFFC_EXTRA.zip)

Sonlen1414, 2017-12-18:
Tested with my digital version of the game, could not reproduce any crashes doing what you said to do. I would reformat your SD as that is the most likely cause of this issue, and not Luma itself.

closed 2017-12-18 by Hikari-chin