LumaTeam/Luma3DS #975: Youtube keeps crashing

RandomMan25XX, 2017-12-19:
System model:

2ds

SysNAND version (+emu/redNAND version if applicable):

11.6.0-39U

Entrypoint (How/what you're using to boot Luma3DS):

Boot9Strap

Luma3DS version:

8.1.1

Luma3DS configuration/options:

Default EmuNAND: 1

Screen brightness: 4

Splash: On

PIN lock: Off

New 3DS CPU: Off
Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

Explanation of the issue:

Upon playing a video that has pre-roll ads the 2ds in question crashes

Steps to reproduce:

Try to load a video known to have pre-roll ads

Launch the application.

Dump file:

[crash_dump_00000004.zip](https://github.com/AuroraWright/Luma3DS/files/1571230/crash_dump_00000004.zip)

ctrninja, 2017-12-20:
That might actually be a software bug and not Luma3DS.

Sonlen1414, 2017-12-20:
Tested on 3 different devices, could not reproduce. Also the Youtube app is notoriously buggy as is, chances are Luma just catches the would be anyways crash.

closed 2017-12-20 by Hikari-chin

RandomMan25XX, 2017-12-20:
Not my device anyways, I couldn't care about the youtube app personally