LumaTeam/Luma3DS #977: 9.0 update breaks homebrew

BlazeUchiha420, 2017-12-20:
Every homebrew app I launch freezes my entire console. This worked until I updated to the latest release. I think I'm going back 8.1.1 for now

takeshineale128, 2017-12-20:
home-brew worked on mine old 3ds xl

BlazeUchiha420, 2017-12-20:
none of them are working on mine. I can launch the the homebrew launcher just fine, but none of my apps are working. They just freeze the console

CodyMKW, 2017-12-20:
homebrew apps seem to work fine on my o3ds with Luma 9.0

BlazeUchiha420, 2017-12-20:
going in through the homebrew launcher app allows them to boot up, then they kick me back out to the homebrew launcher

BlazeUchiha420, 2017-12-20:
it only freezes when I use the app I injected hb launcher into with rosalina

Sonlen1414, 2017-12-20:
The update does NOT break any homebrew apps, redownload the apps or reformat your SD, I just used Download Play and Rosalina Homebrew Injection and managed to load homebrew apps without a problem.
This isn't a Luma issue.

916253, 2017-12-20:
Closed. This isn’t an issue with luma; id advise ensuring all of your homebrew is up to date and that you are using the latest version of hbmenu

closed 2017-12-20 by 916253

BlazeUchiha420, 2017-12-20:
I am using the latest hb menu and they were working perfectly fine before I updated. I used all of my homebrew just earlier today and didn't have this issue until I updated luma. All apps work perfectly fine from my home menu