LumaTeam/Luma3DS #981: Rosalina Restart Buggy / Crashes / Inconsistant

joo456, 2017-12-22:
**System model:**
n3ds

**SysNAND version (+emu/redNAND version if applicable):**

11.6
**Entrypoint (How/what you're using to boot Luma3DS):**

b9s
**Luma3DS version:**

9

bla bla i dont think the config is needed here

**Explanation of the issue:**
rosalina's restart sometimes has makes the bottom screen go all glitchy and sometimes gives out an error (i think it was an errdisp)
it worked fine in 8.1

**Steps to reproduce:**

1. open rosalina

2. hit that restart button

**Dump file:**
idk

Sonlen1414, 2017-12-22:
Without a dump or more information, this can't be further investigated as it isn't reproducable at all, I have used Rosalina's restart function at least 50 times in the last 24 hours without a single issue.

Carrieshot, 2017-12-22:
I have seen the bottom screen go glitchy once earlier for me

Hikari-chin, 2017-12-22:
The bottom screen glitches are to be expected, but I've yet to be able to reproduce any crashes from this option.

closed 2017-12-22 by Hikari-chin