LumaTeam/Luma3DS #982: Luma3DS Crash in DQM Joker 3 [Translated]

leonardobsjr, 2017-12-24:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

Sys 11.6.0-39U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

9.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

It's quite easy to reproduce. In a part of this game, i open this one chest and it crashes. Always crashes. I've gotten the CIA from **ThatSite** and the translation file too. Installed the update according to instructions (DQM J3 is 1.1).

I don't know how can i provide the files i used (CIA, Translation and Save), so i'll just provide the crash dump.

**Steps to reproduce:**

1. Load the Game

2. Try to open the chest.

**Dump file:**

[crash_dump_00000003.zip](https://github.com/AuroraWright/Luma3DS/files/1584570/crash_dump_00000003.zip)

leonardobsjr, 2017-12-24:
Just to add to it, it's reproducible in 8.1.1 too.

Hikari-chin, 2017-12-24:
That being the case, it is extremely likely that this is a big with the translation itself, not Luma.

closed 2017-12-24 by Hikari-chin

leonardobsjr, 2017-12-24:
You're right, it's a text overflow issue.