LumaTeam/Luma3DS #988: Exception Error Homebrew

DeTorre, 2017-12-26:
**System model:**
Old 3ds
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
10.2
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Never got to this point (boot.firm required to power up 3ds from luma3ds)
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( x) (required in order to run, only works with SD card inside)

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( x)

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1586692/crash_dump_00000000.zip)

Zemogiter, 2017-12-26:
I've also got an exception when launching Homebrew Launcher. New3DS firmware 11.6, Luma version 9.0, boot9strap, no emunand, screen brightness 4, splash is off, pin lock is off, new 3DS CPU is set to Clock+L2, game pathing and showing NAND string in system settings is enabled, everything elese is disabled.

Hikari-chin, 2017-12-26:
Tested with 11.6U sysNAND, 11.6J emuNAND, was able to use the Homebrew Menu without issue. Make sure you're using version 2.0 of the HBM and aren't using an incompatible takeover title like H&S. Additionally, please go into the config menu and read the bottom screen description of the "Disable ARM11 Exception Handlers" option before you enable it while reporting an issue.

closed 2017-12-26 by Hikari-chin