LumaTeam/Luma3DS #992: Exception Occurred: Data Abort on Opening Sun

Mags491, 2017-12-29:
Old 3DS

Sys 11.6.0-39E

Boot9Strap

Luma3DS 9.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception: ( )

When I try to open a randomised version of Pokémon Sun on my homebrewed Old3DS, the game fails to open and sends me to a Luma (9.0) Black error screen detailing an exception called Data Abort. I can’t currently dump the file, but I can if it’s absolutely necessary. I followed the same steps precisely that I did for my Omega Ruby randomised version, and that boots and works perfectly. Any and all help is appreciated. I feel it has something to do with Sun being a game that takes up a large amount of space, so starting with that in mind, what would you guys think I should do?

Sonlen1414, 2017-12-29:
This issue is always either a failed randomization, files in the wrong spot, or corrupted data. Nothing to do with Luma. Also make sure you have the updated code.bin and such and not the one from the un-updated base game.

Mags491, 2017-12-29:
Thanks for the quick response. What steps do you think I should take to rectify these potential issues?

Mags491, 2017-12-30:
I decided to try downloading the promoted update (I thought doing so would bust the game in truth), and it must’ve applied the most recent .bin that you mention since it’s working perfectly now. Thanks much.

closed 2017-12-30 by Hikari-chin