LumaTeam/Luma3DS #997: compiling error???

Shadow322, 2018-01-02:
What do i update?

c:\Build_Luma3DS\Luma3DS\sysmodules\rosalina/source/menu.c:199: undefined reference to `MCUHWC_GetBatteryLevel'
collect2.exe: error: ld returned 1 exit status
make[1]: *** [build/rosalina.elf] Error 1
make[1]: Leaving directory `/c/Build_Luma3DS/Luma3DS/sysmodules/rosalina'
make: *** [build/rosalina.cxi] Error 2

Hikari-chin, 2018-01-02:
See 522f105, you need (at least) [ctrulib commit 3bbd2dc](https://github.com/smealum/ctrulib/commit/3bbd2dca163042de947e3c823db227f1d79906d2).

closed 2018-01-02 by Hikari-chin

Shadow322, 2018-01-02:
@Hikari-chin um im lost now do i compile it or what and where do i add it to my environment?

Hikari-chin, 2018-01-02:
Clone the latest ctrulib, `cd ctrulib/libctru`, `make`, then `make install` (`sudo make install` may be necessary if your environment is located in a privileged area).

Shadow322, 2018-01-02:
thanks just a little nudge in the right direction is all i needed