LumaTeam/Luma3DS #999: Luma crashes while Launching Luma Updater... and there is no error logs?

IncognitoJoe, 2018-01-05:
**System model: n3DS**

**SysNAND version: 11.6.0-39U**

**Entrypoint (How/what you're using to boot Luma3DS): B9S**

**Luma3DS version: Forced Replace of boot.firm to 9.0.7**

**Luma3DS configuration/options: **None Selected****

Default EmuNAND: (N/A)

Screen brightness: (5)

Splash: (Y)

PIN lock: (N)

New 3DS CPU: (Auto (primarily 807MHz))

--

Autoboot EmuNAND: (N/A)

Use EmuNAND FIRM if booting with R: (Y)

Enable loading external FIRMs and modules: (N)

Enable game patching: (N)

Show NAND or user string in System Settings: (N)

Show GBA boot screen in patched AGB_FIRM: (N)

Patch ARM9 access: (N)

Set developer UNITINFO: (N)

Disable ARM11 exception handlers: (N)

--

**Explanation of the issue:**
Everything was fine but when I am updating Luma3DS via Luma3DS updater... Luma's boot.firm crashes, or it shoots me back in the Home Screen

**Steps to reproduce:**

1.Launch LumaUpdater

2.Failed?

**Dump file:**

Shadow322, 2018-01-05:
This has nothing to do with Luma3DS in anyway someone close this plez

Hikari-chin, 2018-01-05:
LumaUpdate is not maintained by the Luma3DS team; it is a third party tool developed by Hamcha/KunoichiZ.

closed 2018-01-05 by Hikari-chin