LumaTeam/Luma3DS #1001: ErrDisp pokemon ultrasun

Danksi, 2018-01-06:
With luma 8.1.1 and boot9 1.3 I get the following crash black screen error while playing pokemon ultrasun (the positive thing is that i've got it only twice, today, and yesterday in the morning):

An error occurred (ErrDisp)

Error type: generic
Process ID: 38
Process name: momiji
Process title ID: 0x00040000001b5000

Adress: 0x00346e74
Error code: 0xo96045b

Press any button to reboot

I've started getting it since I used GodMode9 to reset the NNID of my old 3ds with CFW, since I updated ultrasun to 1.1 and since I got Pokemon Bank.

**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

sora10pls, 2018-01-06:
>Enable game patching: (Yes)

Do you have any CROs from 1.0 on your SD Card being loaded through Luma LayeredFS? What about `code.bin`? If there's a mismatch between 1.0 and 1.1 with either of those, you're guaranteed a crash.

Danksi, 2018-01-06:
I'm sorry Sora, but I dont understand what you're saying. This happened me  after I updated ultrasun (physycal cartridge) to 1.1 so maybe the error is related to this

I would try what you say though, but I dont know what a CRO or LayeredFS is

Hikari-chin, 2018-01-06:
This is almost always a result of a damaged SAD card or running un-updated mods such as the No-Outlines Patch on versions it doesn't support.

closed 2018-01-06 by Hikari-chin

Mucephie, 2018-07-17:
I've got this same error, same error screen information, running Luma 9 and sysnand 11.7.0 , SD card is fine, no mods at all. Anyone have a clue ? Sometimes I get it before I even load the save(example: title screen)