LumaTeam/Luma3DS #1002: Ultra Sun Game Crash

realStutz, 2018-01-07:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
I was playing Pokemon Ultra Sun on an O3DS running LumaCFW version 9.0.0 with Boot9Strap, using the default SysNAND, when a fatal error occurred. Details are included below.

An error occurred (ErrDisp)

Error type:              generic

Process ID:             38
Process name:        momiji
Process title ID:      0x00040000001b5000

Address:                0x00346e74
Error code:             0xc88044ab

**Steps to reproduce:**

1. Go to Memorial Hill on Akala Island

2. Challenge the old man with the Jolteon

**Dump file:**

Shadow322, 2018-01-07:
Fill out the forum u was given

Hikari-chin, 2018-01-07:
Can't reproduce, but these sorts of errors are almost always due to corrupted save or game data.

closed 2018-01-07 by Hikari-chin

realStutz, 2018-01-09:
kthx. I'll make sure 2 try a clean save file. kthxbye.

d4mation, 2018-01-10:
This happens intermittently for me with Ultra Moon and Mario & Luigi: Superstar Saga DX. It most often happens to me when returning from Sleep Mode.

I've found reseating my SD Card seems to mitigate the problem at least for a while. I'm wondering if it is caused by a failed read of SD Card contents? I've been meaning to try using a different SD->Micro SD Card adapter to see if it solves it for me as the contacts are looking pretty banged up on the one I'm using.

realStutz, 2018-01-13:
Thanks, I figured something like that was responsible.