LumaTeam/Luma3DS #1003: "An Exception Occurred" Arm 11 error with Smash 3DS

mehmeh1, 2018-01-10:
**System model:**
Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39U SysNAND
**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap/Sighax
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
9.0 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: ( 1)

Splash: (off )

PIN lock: (off )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: (this doesn't show up )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Whenever I load my smash cart, it brings me to an error message saying "An exception occurred", "Processor: ARM11 (core 1)"
"Exception type: prefetch abort (svcbreak)"
"current process: loader (a bunch of 0s)"
"Fault status: Debug event"

What can I do? It has no mods, the only DLC downloaded was the cloud+midgar bundle, and that was through normal means, IDK why the game suddenly stopped working. o3DS

**Steps to reproduce:**

1.

2.

**Dump file:**

mehmeh1, 2018-01-10:
[crash_dump_00000044.zip](https://github.com/AuroraWright/Luma3DS/files/1620416/crash_dump_00000044.zip)
Sorry, forgot the dump
also, steps to reproduce:
Turn on 3ds
Boot smash (cart)
error message

hairareshai, 2018-01-11:
Hi.i have the same error on my old2DS.I Followed the same tutorial as you. same story with some games like Terraria  or Pokemon red ,but some games work ( Tetris and loz four swords anniversary edition).SysNAND version (+emu/redNAND version if applicable):

11.6.0-39U SysNAND
Entrypoint (How/what you're using to boot Luma3DS):
Boot9Strap/Sighax
[e.g. Boot9Strap/Sighax, etc.]

Luma3DS version:
9.0 stable
I need help too!!

If you find something pls let me know .Thx :))

hairareshai, 2018-01-11:
[Crash__Dump.zip](https://github.com/AuroraWright/Luma3DS/files/1624133/Crash__Dump.zip)

SmackBro, 2018-01-12:
Everything seems fine on my end:

n3dsxl
SysNAND 11.6.0-39E
B9S 1.3
Luma 9.0 Stable

I can launch every game on my SD card or their cardridge equivalents (plus various VC injects) without any problem (besides corrupted sprites or audio crackles on romhacks,that shouldn't run on the offcial VC),so this could be an issue with you sd cards? (I'm no expert,just assuming this as installed and cardridge titles work fine on my end)

hairareshai, 2018-01-12:
Smackbro ,I would appreciate if you could give me a link from where you download your.cia games.are you using free e shop /villain?

SmackBro, 2018-01-12:
I didn't pirate any of my "digital" games, I simply used godmode to dump them from my cardridges or bought them from the e-shop.

closed 2018-01-12 by Hikari-chin

mehmeh1, 2018-01-14:
why was this closed?

SmackBro, 2018-01-14:
@mehmeh1  Because I proved that this is no bug that occures on every console. (unlike the bugs that occure with input redirection,or when you start the debugger in rosalina menu two or three times).
So this issue has to be an external issue,your SD might be the cause of your problem,you should ask for help on reddit (/r/3dshacks) if you didn't do this already,as this is no support forum. This is an issue tracker,so you submit your found bug or suggest a feature and a "mod/dev" will take a look on that issue,if it is invalid like your's it will get closed,as this is no direct bug inside Luma.

mehmeh1, 2018-01-14:
ok, thanks