LumaTeam/Luma3DS #1006: system crash loading physical game cart (ultra sun)

vihoremr, 2018-01-16:
**System model:** New 3DS

**SysNAND version (+emu/redNAND version if applicable):** 11.6.0-39U SYSNAND

**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap

**Luma3DS version:** 9.0

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: ( 4)

Splash: ( Off)

PIN lock: ( Off)

New 3DS CPU: ( Off)
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

System crashes when trying to launch a physical game cart (Pokemon Ultra Sun). This is similar to #1003 

Backstory:
I followed 3ds.guide back in the A9HL days, on firmware 11.3. No problems. Moved to b9s and Luma 8+ and didn't have a problem either. Bought Pokemon Sun, then Pokemon Ultra Sun and didn't have problems. I found out about Anemone3DS and 3DShell so I installed them using FBI. I attemped to install a theme using Anemone3DS but my system didn't change themes so I figured it didn't work and left it alone. After a few days of playing Ultra Sun without a problem I started getting crashes when trying to play that cart. I tried my regular Pokemon Sun cart and it works fine. TWLoader and DSTT work fine. I'm unable to start any other .cias or this Ultra Sun cart. 

**Steps to reproduce:**

1. Turn on 3DS 

2. Try to launch physical game cart (Pokemon Ultra Sun)

3. Crash

**Tried:**

1. Using a new SD card and following https://3ds.guide/finalizing-setup

2. Make sure everything (FBI 2.4.14, godmodev9 1.5.1, b9s 1.3, Luma 9) is up to date

3. Delete home menu ext data

**Dump file:** [crash_dump_00000045.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1637144/crash_dump_00000045.dmp.zip)

SmackBro, 2018-01-17:
Do you have any Luma Game Patches for Pokemon Ultra Sun active,the no Outlines patch will cause a crash on 1.1 and I don't know if it got updated already.

Hikari-chin, 2018-01-19:
Can't reproduce this; you've likely got corrupted saved data, patches incompatible with your Ultra Sun version, or some other similar problem.

closed 2018-01-19 by Hikari-chin