LumaTeam/Luma3DS #1015: ARM 11 Prefetch abort (kernel panic)

lincruste, 2018-01-22:
**System model:

New 3DS XL

**SysNAND version :
11.6.0-39E SysNAND

**Entrypoint :
Boot9Strap/Sighax

**Luma3DS version:**
v9.0 (stable)

**Luma3DS configuration/options:**

Show NAND or user string in System Settings: (X )

**Explanation of the issue:**

When using the console, I randomly encounter a kernel panic issue.
ARM 11 Prefetch abort (kernel panic)

**Steps to reproduce:**

1. Play Binding of Isaac for N3DS or a homebrew game like Retroarch

2. Encounter the bug randomly

**Dump file:
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1650399/crash_dump_00000000.zip)
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1650400/crash_dump_00000000.zip)

cheatfreak47, 2018-01-22:
this is likely just Luma 3DS catching a crash in whatever homebrew or 3DS game you are playing and displaying a verbose error instead of just crashing and rebooting the console with no information provided.

Binding of Isaac in particular is really quite buggy and crashes often.

Hikari-chin, 2018-01-22:
Can't reproduce this.

closed 2018-01-22 by Hikari-chin

6100m, 2018-05-10:
Nuh uh. I fucking can reproduce this. It happens mostly when one of my downgrades fails. Usually restoring a NAND backup fixes that right up :)

P.S. @lincruste sorry if @Hikari-chin is always rude you know how the chinese are :/