LumaTeam/Luma3DS #1019: Mario & Luigi SSS+BM, Earthbound (VC) and 3D Sonic the Hedgehog 2 crash on boot.

Takatsuki, 2018-01-25:
**System model:**

New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9strap

**Luma3DS version:**

9.0

**Luma3DS configuration/options:**

Everything is default.

**Explanation of the issue:**

There's actually 3 games that I've found to cause the system to crash when trying to boot them (on the Nintendo 3DS loading screen), though I already deleted one of them from my system so I was unable to get the crash dump of that one (the game in question is 3D Sonic the Hedgehog 2).

However, Mario & Luigi: Superstar Saga + Bowser's Minions, as well as Earthbound (Virtual Console) cause the system to crash during the above stated loading screen when trying to load them from the home menu. I am given an exception error and a crash screen. I've included crash dumps for M&L:SSS+BM and Earthbound, as well as screenshots of the error codes. 

I've tried roughly 30 other games and they all work flawlessly so I don't think that this is a general issue (such as the MicroSD or something), it's most likely an issue specific to these 3 games.

**Steps to reproduce:**

1. Install these 3 games from the MicroSD using FBI (install and delete cias).

2. Simply try and boot the game from the home menu and it will crash shortly after the "Nintendo 3DS" loading screen shows up. This happens every single time without fail. 

**Dump file:**

[Luma arm11 Crash Dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1665223/Luma.arm11.Crash.Dumps.zip)

Screenshots: https://imgur.com/a/vc4kX 
 
In other news, I think I need to clean my system...

Hikari-chin, 2018-01-25:
The CIAs you installed were corrupted/damaged in some way.

closed 2018-01-25 by Hikari-chin

Takatsuki, 2018-01-25:
They're not corrupted. They were downloaded directly from Nintendo's servers.

Sonlen1414, 2018-01-25:
If they were downloaded from Nintendos Servers, you would have gotten them from the eShop and not had to use FBI, they are clearly not from the eShop and clearly are corrupted.

piratesephiroth, 2018-01-25:
I'm not sure why you even suggest that the CIAs are corrupted. If that was the case then they wouldn't even install because Luma doesn't bypass any integrity checks.

It's certainly because he used eshop-encrypted CIAs so he doesn't have the seed. He just has to go to "Titles" in FBI, select the problematic game and import the seed from Nintendo's servers.