LumaTeam/Luma3DS #1022: HELP! (An exception occurred.  processor:  ARM11 ((core 0)). exception type: prefetch abort .  fault status: Translation selection  menu((000400300008f02))

jboone321, 2018-02-03:
I'm trying to load soundhax on my 3ds 9.6,0-10U.   When I boot it, it goes to luma 3ds configuration. 

here are the settings

Autoboot EmuNAND: (x)

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

Afterwards, I go on Nintendo 3ds Sound and go on nedwill. After that I get the error. Please HELP!
[crash_dump_00000000.dmp.zip](https://github.com/Auror
[crash_dump_00000000.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1691213/crash_dump_00000000.dmp.zip)
aWright/Luma3DS/files/1691210/crash_dump_00000000.dmp.zip)

Theres the crash dump if needed.

Hikari-chin, 2018-02-03:
You can't use soundhax with Luma v8.0 or higher, you must use Rosalina's takeover title method.

closed 2018-02-03 by Hikari-chin

jboone321, 2018-02-03:
Thank you. 

*but who the f--- said you can close this thread*

Hikari-chin, 2018-02-03:
I did.