LumaTeam/Luma3DS #1023: OK. I've been to a method on how to fix this. its called Rosalinda takeover title method, but I need someone to elaborate on what that is?

jboone321, 2018-02-03:
I'm trying to load soundhax on my 3ds 9.6,0-10U. When I boot it, it goes to luma 3ds configuration.

here are the settings

Autoboot EmuNAND: (x)

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

Afterwards, I go on Nintendo 3ds Sound and go on nedwill. After that I get the error. Please HELP!
[crash_dump_00000000.dmp.zip](https://github.com/Auror
crash_dump_00000000.dmp.zip
aWright/Luma3DS/files/1691210/crash_dump_00000000.dmp.zip)

Theres the crash dump if needed.

jboone321, 2018-02-03:
FRAP that's supposed to say *told a method.

jboone321, 2018-02-03:
*crap.

ihaveamac, 2018-02-03:
you know you can edit your issue title and posts, right? especially given the issue body here is completely unrelated to the title.z

Hikari-chin, 2018-02-03:
The issue tracker isn't for tech support.

closed 2018-02-03 by Hikari-chin

jboone321, 2018-02-03:
Hikari STOP CLOSING SHIT.

ihaveamac, 2018-02-03:
go somewhere else for tech support.