LumaTeam/Luma3DS #1030: Arm11 Crash

AtSburbsEnd, 2018-02-08:
**System model:**
New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

 11.6.0-39U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

9.0-2a840f2

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )
(I have no idea about any of these) 
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )
(I don't know ow about these either) 
--

**Explanation of the issue:**

So I have two different (maybe three) issues that come up with Arm11. 
I have Pokemon Moon and Ultra moon, and in both games when I go to the festival plaza, and use the dying station, I get an arm11 crash. 

The other is in minecraft. If I change any of my world's settings then attempt to open the world, I get a crash. Sometimes it goes through without crashing on the load, but then crashes later. 

**Steps to reproduce:**

1. Go to festival plaza in moon/ultra moon

2. Attempt to use the dye tent

OTHER ISSUE

1. Open minecraft
2. Edit world (in my case I made a creative world go from easy to peaceful) 
3. It crashes while loading, or not long after it has loaded. 

**Dump file:**

You'll have to excuse me on this. I'm working off my phone for this stuff right now so the most I can do is get a picture of my 3ds screen when the crash happens.

AtSburbsEnd, 2018-02-08:
[image]
This is the crash in Minecraft

Sonlen1414, 2018-02-09:
Most likely corrupted game data or a SD issue. Back up your SD, reformat it, restore it, if needed use the eShop to repair any game data, if it is a cart based game you may need to backup the Extdata and at delete it and see if that fixes it.

AtSburbsEnd, 2018-02-09:
Alright I'll try that. Sorry I couldn't get the other screenshots, personal stuff came up. I'll get back to you with my results

closed 2018-02-09 by Hikari-chin