LumaTeam/Luma3DS #1033: Compiling Luma3DS 9.0 - boot.firm exception

limbonic1, 2018-02-11:
**System model:**
New 3DS XL
**SysNAND version (+emu/redNAND version if applicable):**
11.6.0-39E SysNAND, 11.3.0-xxE EmuNAND]
**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
**Luma3DS version:**
9.0
(https://github.com/AuroraWright/Luma3DS/commit/76d274cfe27d7a5a41d8094fe51ccb1fe2ab15e0)

**Luma3DS configuration/options:**
Default EmuNAND: (1)
Screen brightness: (4)
Enable game patching: (x)
Show NAND or user string in System Settings: (x)

--

**Explanation of the issue:**
Compling via make firm succeeds without any errors or warnings.
Starting the 3DS shows the setup screen like normal, after pressing start
it shows this (the chainloader works):

An exception occured
ARM11 (core1)
data abort
Permission - page
sm (0000000000000000)

**Steps to reproduce:**

1. make firm
2. copy over boot.firm
2. power on 3ds

**Dump file:**

Shadow322, 2018-02-11:
why such an old commit?

limbonic1, 2018-02-12:
It's the third last commit, but i should have asked the other way arround.
Since Luma 9.0 none of my own builds work as written above.

Everything is up to date in my build environment and i never had problems
compiling Luma 6, 7 or 8. Even payloads like godmode9 still work.

It is just a strange issue, make firm throws no errors.

Shadow322, 2018-02-12:
then sounds like your environment is messed up either installed wrong or outdated some commits back the environment needed updated to compile right

Hikari-chin, 2018-02-12:
Can't reproduce, your build environment is likely out of date or otherwise incorrect.

closed 2018-02-12 by Hikari-chin

piratesephiroth, 2018-02-12:
> i never do this make firm command u referring to
make firm is just the default option, you don't really need to use it