LumaTeam/Luma3DS #1034: LayeredFS game patching not working with USA Tomodachi Life

0ICED0, 2018-02-12:
I have attempted to patch my USA version of Tomodachi Life with a music hack. When I select the option "Enable game patching" with all my files in G:\luma\titles\(Tomodachi Life's title id)\romfs\ nothing happens. All of the files I wish to patch are in the right places but nothing would load. I have attempted this on other regional versions of the game and have only had success on the Korean version of the game. (Attempted on Luma 9.0)

0ICED0, 2018-02-12:
The console I am using is a North American New 3ds XL

AuroraWright, 2018-02-13:
Can you send me the files you modified?

0ICED0, 2018-02-13:
@AuroraWright What service should I send you the files on?

AuroraWright, 2018-02-13:
You can upload them to transfer.sh or similar

0ICED0, 2018-02-13:
(Hopefully I am doing this right) https://transfer.sh/BKksW/Clone_map_day02a.dspadpcm.bcstm replaces the daytime music outside, it goes in \sound\stream. I have tested this by rebuilding the game and it works as intended but rebuilding is icky and slow.

Shadow322, 2018-02-13:
Fat32 is all the 3ds can read LOL

closed 2018-07-25 by yuukieve