LumaTeam/Luma3DS #1035: No Home Menu :c

carlosg269, 2018-02-16:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

sysnand 11.6

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

9.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I already saw some similar issues but i dont have seen the same that is happening to me, it doesnt let me boot into the home menu, i still have access to the chainloader, and when ARM11 exception handlers are disabled it stays in black screen, but when i press the power off it sends me the sleep mode screen and when i press the home menu button to go back it stays in that white screen

**Steps to reproduce:**

1. I dont know

2.

**Dump file:**

[image]

Hikari-chin, 2018-02-16:
Can't reproduce; it's likely you've got some corrupted data somewhere. Try clearing your Home Menu extdata.

closed 2018-02-16 by Hikari-chin

carlosg269, 2018-02-16:
I've already resolved on the discord group, it's a matter of deleting the data folder in the sysnand ;)