LumaTeam/Luma3DS #1036: An exception occurred

Supermariotyt, 2018-02-20:
**System model:Old 2ds

**SysNAND version (+emu/redNAND version if applicable):** Unacapable

**Entrypoint (How/what you're using to boot Luma3DS):** Soundhax/Boot9strap

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:** 9.0 (cannot view in system settings)

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (Impossible to configure )

Screen brightness: ( impossible to configure)

Splash: (Impossible to configure )

PIN lock: ( Impossible to configure)

New 3DS CPU: ( )

--

Autoboot EmuNAND: (No )

Use EmuNAND FIRM if booting with R: ( no)

Enable loading external FIRMs and modules: ( No)

Enable game patching: (Impossible to configure  )

Show NAND or user string in System Settings: ( Yes)

Show GBA boot screen in patched AGB_FIRM: ( no)

Patch ARM9 access: ( ?)

Set developer UNITINFO: ( no)

Disable ARM11 exception handlers: (no )

--

**Explanation of the issue:** An exception occurred
Processor: Arm11 (core 1)
Exception type:Data abort
Fault status: Translation-Page
Current process: Menu (000400300008f02)
Ro 00000000 R1 00000000
r2 fff8c424 r300000002
r4 deadbabe r5 0010201c
r6 deadbabe r7 00000000
r8 00101bc8 r9 00000001
r10 343850e8 r11 oo100fb8
r12 0010r458 sp 35040000
lr 0010201c pc 0010201c
cpsr 60000010 fpexc 00000700
far 3504000 Access type: Read 

Idk the steps to reproduce, I just want this fixed. 
No pictures, but i have 3 dump files.

ValliantStorme, 2018-02-20:
Are you not holding START on boot to view luma settings? And what do you mean by "Soundhax/Boot9strap", Soundhax is incompatible with Luma 9.0.

Supermariotyt, 2018-02-20:
So Are you staying I should download luma3ds 8.11?

Shadow322, 2018-02-20:
Soundhax is patched since 11.4

Hikari-chin, 2018-02-20:
Even if soundhax hadn't been patched in 11.4, you still can't use it under modern versions of Luma because of memory map changes with Rosalina. You must use the takeover title mechanism instead.

closed 2018-02-20 by Hikari-chin

Supermariotyt, 2018-02-21:
I think you are all wrong. The 3ds guide website says that you can only use soundhax to boot into luma. And also, Im on version 11.3, not 11.4

Qyriad, 2018-02-21:
Soundhax is a userland arbitrary code execution exploit. It alone cannot boot into Luma. There's also no point to using Soundhax if you already have custom firmware.

Supermariotyt, 2018-02-22:
THEN PLEASE TELL ME WHAT TO DO. THIS ISNT A "BE A SMARTASS" FORUM, ITS A HELP FORUM. TELL ME IF I NEED TO DELETE ANYTHING.

Hikari-chin, 2018-02-22:
Actually, this is an issue tracker. Not a "help forum".