LumaTeam/Luma3DS #1038: Game patches load only from boot drive

Localhorst86, 2018-02-22:
**System model: New 3DS**
**SysNAND version: 11.6.0-39E**
**Entrypoint (How/what you're using to boot Luma3DS): Boot9strap**
**Luma3DS version: 9.0 stable**

When launching Luma3DS from CTRNAND, game patching (like, for example, LayeredFS) will not work if they are stored in the luma folder of the SD card. In fact, the same happens with any external files loaded by luma (payloads, for example).

By placing the luma boot.firm on the root of the SD card the paches and layeredFS will load but I was wondering if this is indeed the intended behaviour or if this is an oversight. Maybe that has already been considered and there is already a hidden setting for people that would like to keep their system "self-contained" - If so, that setting needs to be documented somewhere.

piratesephiroth, 2018-02-23:
hmm yeah, it's clearly an oversight

Localhorst86, 2018-02-23:
Yeah, the main concern would be patches as they can be quite large and space on CTRNAND is limited. However, I believe that it would not be a bad idea to include payload chainloading from SD card as well as to disencourage people to actively put files (other than boot.firm) on the CTRNAND.

Shadow322, 2018-02-23:
Just run boot.firm from sd card u will never get the team to fix this I have been told this on here so its fruitless to even bother

Hikari-chin, 2018-02-23:
This has been reported before, and as said then, it is intended behavior.

closed 2018-02-23 by Hikari-chin

Localhorst86, 2018-02-24:
Gotcha, putting patches on ctrnand then.