LumaTeam/Luma3DS #1047: HELP I CANT LOAD HOMEBREW

ShinyBagonHunter, 2018-03-15:
**System model:**
2ds
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.2.0-35U SysNAND, 11.2.0-35U EmuNAND]
11.2.0-35U
**Entrypoint (How/what you're using to boot Luma3DS):**
Homebrew
[e.g. Boot9Strap/Sighax, etc.]
soundHax
**Luma3DS version:**
9.0
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I cannot load homebrew after I installed luma cfw

**Steps to reproduce:**

1. ?

2.

**Dump file:**

[crash_dump_00000000.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1817051/crash_dump_00000000.dmp.zip)

ShinyBagonHunter, 2018-03-15:
Please respond ASAP

Sonlen1414, 2018-03-15:
You can't use soundhax after you get Luma running, you need to use Rosalina, I recommend you read https://3ds.hacks.guide/finalizing-setup to see how it works easily.

fincs, 2018-03-15:
This is not a support forum.

closed 2018-03-15 by fincs