LumaTeam/Luma3DS #1051: TWLoader is broken in the latest commit (1182d3a).

nl255, 2018-03-24:
When trying to run any DS game through TWloader on the latest commit (1182d3a) it gives an arm9 prefetch abort (svcBreak) error.  It works fine on older versions such as 4f54596 or 938cb64. 

[crash_dump_00000001.zip](https://github.com/AuroraWright/Luma3DS/files/1844530/crash_dump_00000001.zip)

SpongebobPlace, 2018-04-05:
be sure to report this on TWLoader's GitHub page if you haven't

https://github.com/Robz8/TWLoader/issues

nl255, 2018-04-05:
After some more testing including trying SRLoader (which crashed immediately) and even a few DSIware games I have found the issue is not just with TWLoader but anything that runs in DSi mode.  As such I am closing this issue in favor of #1058

closed 2018-04-05 by nl255