LumaTeam/Luma3DS #1055: joy con hacks brocked me!!!!!!! and my switch to!!!!

A42MphTortoise, 2018-04-01:
**System model:** old 3ds color is red and i peeld of the back sticker too

**SysNAND version (+emu/redNAND version if applicable):** 11  6 39 U

**Entrypoint (How/what you're using to boot Luma3DS):** booot 9 strap

**Luma3DS version:** joy con hacks version

**Luma3DS configuration/options:**

Default EmuNAND: (1 but i can set it too 2 or 3 or 4 if you want)

Screen brightness: (1 but i can also change this)

Splash: (after because i like to see my picture of zelda climing the mountain)

PIN lock: (i turned this off after realizig i coudnt change to konome code)

New 3DS CPU: not aplliciable )

--

Autoboot EmuNAND: ( this one is blank)

Use EmuNAND FIRM if booting with R: ( also blank)

Enable loading external FIRMs and modules: ( will i brick worse if i turn this on???)

Enable game patching: (no thats hacking and ill get banned)

Show NAND or user string in System Settings: (this one has an x)

Show GBA boot screen in patched AGB_FIRM: ( i dont know what this does)

Patch ARM9 access: ( is this like that arm 9 loder hacks because i here about that on gba temp)

Set developer UNITINFO: ( i cant make games in game maker studio so im no developer)

Disable ARM11 exception handlers: ( blank it says i cant make an issue but if i cant make an issue then why i am i brorked)

--

**Explanation of the issue:**

i installed joy con hacks but now i receive error 002-0102 when i go online and i cant play any games either but my friend list was turned on and my joy con's cant connect to my switch anymore please help have i bricked i cant afford a new 3ds please help maybe its my twl firm got deleted

**Steps to reproduce:**

1. install joy con hack

2. play games

**Dump file:** where is this on my emunand??? canot find..!

Shadow322, 2018-04-01:
Enough trolling out of u for the night

AuroraWright, 2018-04-01:
It seems like you didn't position the joycons correctly and direct kernel access wasn't 100% achieved...

916253, 2018-04-01:
weird, worked fine for me

CrisMonti, 2018-04-01:
Plz no more april fools!!!

fincs, 2018-04-01:
To me, it sounds like either your computer ran out of wireless RAM when you tried to download the latest version of Luma3DS, or you have read-only wireless RAM installed. This error is also typical of users who lack the wireless RAM adapter required to bind the JoyCons to the kernel. As a workaround, you may try to [download more wireless RAM](https://downloadmoreram.com/), flush and try again.

Dimensional, 2018-04-01:
I tried to do this but the screen only displayed "Error: ID10T"

Conan179, 2018-04-01:
realy nice error code.

closed 2018-04-04 by Hikari-chin

Multimegamander, 2018-08-20:
This should be reopened