LumaTeam/Luma3DS #1058: DSIware arm9 error with the latest commit (1182d3a).

nl255, 2018-04-05:
When trying to run any DSiware game including SRLoader or DS games through TWLoader on the latest commit (1182d3a) it gives an arm9 prefetch abort (svcBreak) error. It works fine on older versions such as 4f54596 or 938cb64.

nl255, 2018-04-05:
Update:  It turns out the problem was with the filename I gave the latest commit.  Apparently it doesn't like 1-Luma-nightly-1182d3a-20180324.firm and the reason the earlier version worked is because it was named 1-Luma-nightly-938cb64.old2.firm which is shorter.

SpongebobPlace, 2018-04-05:
Please be sure to send the TWLoader crash dump to the TWLoader developers also it may have something to do with that.

nl255, 2018-04-06:
It is not a TWLoader issue as it happens with any DSi mode software including games from the eshop like Shantae or Mighty Flip Champs.  Fortunately it can be worked around by renaming the Luma .firm file to something shorter but that limitation should be mentioned in the wiki now that things like fastboot3ds and gm9 bootloader are becoming more popular.

Nanquitas, 2018-04-06:
@TuxSH made some changes that should fix this. (356a94cc2efb9cf36bf6d53d9a44133f283ccbbc)
Mind trying with a recent hourly and reporting back ?

nl255, 2018-04-06:
It works now with my preferred nightly naming scheme (1-Luma-nightly-[commit]-[date].firm) but the path size limitation information should still be added to the wiki in both the instalation and troubleshooting section.

Nanquitas, 2018-04-06:
The maximum is now of 80 characters. I think it's long enough for every scenarios.
Anyway, good if it's solved, you should close the issue. ;)