LumaTeam/Luma3DS #1066: [Feature Request] Amiibo scanning simulation

ederenzi78, 2018-04-17:
I don't know if it's feasible but it would be nice to have a feature to virtually put an Amiibo on the NFC sensor in order to unlock game features without actually using the figurines. In order to do this, Luma could use the ROM inside each Amiibo (which the used should have put inside a specific folder).

Shadow322, 2018-04-17:
never happening as this would support piracy

Dimensional, 2018-04-18:
Not solely, like your suggestion. Disabling Signature checking allows for better homebrew capabilities.

profi200, 2018-04-19:
This is a feature that really only supports piracy as mentioned above. I doubt very much the Luma devs will consider this. There are other ways to archive a similar goal.

urherenow, 2018-04-19:
Piracy isn’t the sole use case. I have a 5ft long window sill packed with Amiibos. When I deploy on a ship, I can’t possibly take them with me. I use thenaya to back them up and flash them to NFC stickers. With a feature like this, I wouldn’t even need to carry (and worry about losing) the stickers.

Nuck-TH, 2018-04-20:
Morality discussion can be pretty much endless.
This feature can be used for convenience, because Amiibos are not things you can easily carry around.
Much like signature spoofing this can be used for piracy as well, but decision is always on the user, let's not decide for them.

Multimegamander, 2018-04-20:
welp, time to wait for a fork with this feature

PikachuR77, 2018-04-30:
Also, this would let Old 3ds users use amiibo without the adapter.

Hikari-chin, 2018-05-28:
It's as @profi200 and @Dimensional say; this won't be implemented by us.

closed 2018-05-28 by Hikari-chin