LumaTeam/Luma3DS #1070: Dillon's Dead Heat Breakers (Import) crashing

MarinaRuby, 2018-04-27:
**System model:**

Old 3ds

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9strap

**Luma3DS version:**

9.0

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: ( 4)

Splash: (off)

PIN lock: (off )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I recently imported the Japanese version of Dillon's Dead Heat Breakers and got the locale.txt for the game set up and everything. Played it just fine for about an hour and some change until I ran into this right before hiring some Miis for a mission/stage.

[deadheatbreakers-crashdump](https://user-images.githubusercontent.com/38794888/39381358-d50f0288-4a2f-11e8-8690-1d002174e591.png)

**Steps to reproduce:**

1. Not sure how to describe this one, but I just tried to get through this text box...
[image]

 and then right after the screen fades while loading the next part, the crash that happened above appeared moments later. Not sure what to do or what the crash means.

**Dump file:**

[crash_dump_00000003.zip](https://github.com/AuroraWright/Luma3DS/files/1956448/crash_dump_00000003.zip)

MarinaRuby, 2018-05-25:
[image]
here's a screen of the crash as well. Again...0 issues on my japanese 3ds but for both english and japanese versions of this game, I'm having this issue.

Hikari-chin, 2018-05-25:
Given that it's functional on your Japanese 3DS but not the English 3DS and the crashes do not always occur in the same section, it's somewhat likely you've got a damaged SD card or something of that nature.

closed 2018-05-25 by Hikari-chin

MarinaRuby, 2018-05-26:
@Hikari-chin 
I went and bought a new SD card and it still happened. Deleted the game, re downloaded it, still happens.

I'm half tempted to buy either the European or Australian version of the game and want to see if I get the same result because the retail Japanese version gives me the same issue. Any other trouble shooting tips I could possibly try out.

One other thing I need to reiterate : If I clear a stage on my japanese 3ds and bring the retail game back to my main 3ds, I can load the stage that crashed just fine. What's driving me nuts is that the japanese 3ds loaded everything from scratch of the game until the end of the game. But now after I buy the localized version of the game from the eshop it's giving me this issue again.

I don't know if it's any extdata, I don't know if it's because of the Miis or loading the new map or what. I'm half tempted to just remove b9s and luma to see if it still happens or just doesn't launch.

Hikari-chin, 2018-05-26:
If you copied your data from the old SD card to the new, you could've copied over damaged data. Try starting from scratch. 

As a last resort, you could try restoring your last known good NAND backup or clean ctrtransfer image.

MarinaRuby, 2018-05-26:
Just to be sure, you want me to format my sd card on pc and redownload the game from the eshop?

Hikari-chin, 2018-05-26:
Yes. But be aware that all your saved data for every game you own will be lost by formatting.

MarinaRuby, 2018-05-27:
@Hikari-chin 
Sorry for the wait, had work the night before and yesterday....anyway it crashed again even after formatting the card and downloading it again. This time....before the 2nd stage.
[image]
I'm seriously lost, this is the only game in last 6+ years with this system  that I've had a problem like this with. The prequels to this game didn't do this, project x zone 2 (english and japanese) didn't do this....it's only this game.

I can't really go back to a nand backup since my last back up was from last year and haven't had much issues nor did I make notes about that last back up. 

I honestly just want to remove b9s and luma just to see if it still happens and just restore it if it still doesn't work with another homebrew method. Unless there's another method I can try.

Dimensional, 2018-05-27:
If that is a JAP import and you're running it on a US 3DS, removing CFW on your system will prevent you from playing it all together, as the console is region locked.

MarinaRuby, 2018-05-27:
@Dimensional 
I absolutely understand that, but the problem is that the US eshop version is giving me the exact same issue as the Japanese version last month. I can't play either version past the beginning stages.

piratesephiroth, 2018-05-28:
what happens if you disable the ARM11 exception handlers?

MarinaRuby, 2018-05-28:
@piratesephiroth 
Upon trying to load the hotel lobby (where it keeps crashing) the game black screens and reboots to the main menu.

Hikari-chin, 2018-05-28:
Further discussion of this can take place in the NH Discord, linked in the issue template comments.