LumaTeam/Luma3DS #1071: [Revision Request] Alter the PIN screen to be more secure?

MA-Designs, 2018-04-30:
Yes.  Hi.  I love Luma3DS.  Thanks.  But I have a request: a change to the PIN unlock page that would make it more secure.

To be perfectly honest: I would prefer the text I added personally to be on the top screen and the enter PIN on the touch but that doesn’t matter.  Maybe give me the option?

My main request is this: Could you remove the first blue line and remove any indication of the number of digits needed to enter.

I know what the code is because I set it, I also know how many characters are in it and how to enter them for the self-same reason, but there’s no reason to share any of that with someone trying to guess it.

I mean, I don’t really care because it’s just my N3DS.  I’m really only worried about theft.  It’s mostly because it’s just bad security.  Keep all details like how many digits and even how to enter them from someone who would break in for as long as possible.  Let them figure it out the hard way.

And one last aside:
This is a bit harder to implement: but would it be possible to add a modifier like
Hold up on the joystick while imputing the rest of the code or holding left and right or touching the screen.  A modifier.

Sorry if this is either in the wrong place or something like that.  Also sorry if you think this is a useless idea: after all it’s just the N3DS, anyway.  I counter that this fact is not a good enough excuse for poor security measures.  And what we have now is, in my opinion, simply bad form.

WubTheGame, 2018-04-30:
I think it would be better for the removal of the blue line should be optional, because if you haven't used your 3DS in a while and have changed PINs multiple times along with number of buttons to press, you would have more trouble getting into your system.

I may be wrong, but i'm unsure of whether or not you could pull up the keyboard because i think it's part of the OS, which at the time of inputting your PIN, isn't running yet. However perhaps there could be a keyboard like in Godmode9, but with a layout from a typical keyboard or Digital 3DS keyboard. And of course it would be touch screen, or controlled with the D-Pad and/or Circle Pad.

And for my last bit, if a person who were to steal your system be smart enough to know how 3DS hacks work, he could easily just put the SD card into a computer and delete the PIN. That is, unless the PIN gets integrated into another file or something (Or if it already has, i don't remember).

Whatever Aurora and the team decides to do is of course up to them. But that's just my two cents.

ev1l0rd, 2018-05-08:
Just one thing to note is that the boot PIN can easily be removed (even if its on the NAND) using GodMode9. Just use it as the main payload instead of luma3ds and remove the files.

MA-Designs, 2018-05-10:
Yes.  I know it can be easily removed.  Like I said before, the thief may not.  Why make anything easy for them.

Hikari-chin, 2018-05-28:
There's really no need for this. This feature was more to prevent children or similar from messing with your device, not as a security measure; it's trivially bypassable.

closed 2018-05-28 by Hikari-chin