LumaTeam/Luma3DS #1075: [FEATURE REQUEST] Splash screen on config menu?

6100m, 2018-05-04:
That would look pretty OP.

Hikari-chin, 2018-05-04:
Duplicate of #940.

closed 2018-05-04 by Hikari-chin

6100m, 2018-05-09:
Well, if the PIN Screen supports it, then it should support the config with a little something like this :)

#include &lt;3ds.h&gt;
#include &lt;sf2d.h&gt;
#include &lt;stdio.h&gt;
 
extern const struct {
 unsigned int width;
 unsigned int height;
 unsigned int bytes_per_pixel;
 unsigned char pixel_data[];
} image;
 
int main(int argc, char **argv)
{
 //Initialize the sf2d lib
 sf2d_init();
 //Set the background color
 sf2d_set_clear_color(RGBA8(0x40, 0x40, 0x40, 0xFF));
 
sf2d_texture *tex_image = sf2d_create_texture_mem_RGBA8(image.pixel_data, image.width, image.height, TEXFMT_RGBA8, SF2D_PLACE_RAM);
 
// Main loop
 while (aptMainLoop())
 {
 //Scan all the inputs. This should be done once for each frame
 hidScanInput();
 
//hidKeysDown returns information about which buttons have been just pressed (and they weren't in the previous frame)
 //hidKeysHeld returns information about which buttons are currently pressed (regardless if they were pressed or not pressed in the previous frame)
 //hidKeysUp returns information about which buttons are not pressed but were pressed in the previous frame
 u32 kDown = hidKeysDown();
 
if (kDown &amp; KEY_START) break; // break in order to return to hbmenu
 
sf2d_start_frame(GFX_TOP, GFX_LEFT);
 sf2d_draw_texture(tex_image, 0, 0);
 sf2d_end_frame();
 //Swap the buffers
 sf2d_swapbuffers();
 }
 sf2d_free_texture(tex_image);
 
sf2d_fini();
 return 0;

Will look more into this :)