LumaTeam/Luma3DS #1076: [BUG] AGB roms with custom intros makes the system freeze?

6100m, 2018-05-04:
Yeah, you know how No-Intro sets are named because the person who made the set doesn't add intros to the rom like some people who rip roms do? Well, believe it or not, but whenever I load a rom that has a custom intro like that, hence using Ultimate 3DS VC GBA Injector, it seems to crash? Not sure why the hell this happens, it doesn't even give me a errdisp, all it does is just freeze. Like, the system just stops completely. No error, no nothing.

piratesephiroth, 2018-05-04:
yeah use GBATA to remove the intro and hopefully they will work

6100m, 2018-05-04:
Well, what I meant was when I enable AGB logo in the config, but never mind, i can just remove the intros. Closing this now.

closed 2018-05-04 by 6100m

AuroraWright, 2018-05-10:
One of the reasons the GBA logo is toggleable is that most of them are improperly made (I think they don't fix a checksum somewhere or something like that) and the GBA BIOS (what runs while you see the logo) won't run them.