LumaTeam/Luma3DS #1079: An exception occured

YummyYoshi, 2018-05-07:
System model:

New 3ds

SysNAND version (+emu/redNAND version if applicable):

11.6.0-39U SysNAND

Entrypoint (How/what you're using to boot Luma3DS):

Boot9strap

Luma3DS version:

9.0

Luma3DS configuration/options:

Default EmuNAND: (1 )

Screen brightness: ( 4)

Splash: (off)

PIN lock: (off )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

Explanation of the issue:

Triforce Heroes used to work, have not played for a couple months, and now it doesnt work. I've deleted and redownloaded, but still not working. I downloaded it on the freeshop. And every other thing I've downloaded works. This is what the crash says:

Processor: ARM11 (core0)
Exception type: data abort
Fault status: Translation - Section 
Current process Alice (000etc.)

Hikari-chin, 2018-05-07:
No support for piracy will be provided here.

closed 2018-05-07 by Hikari-chin

6100m, 2018-05-10:
Uh.. The whole point of CFW is for piracy. PICNIC is on you.