LumaTeam/Luma3DS #1080: [Revision Request] debug mode added to debugging info patch..?

6100m, 2018-05-10:
Just had a thought @TuxSH 

Rerember how the 3ds devkit logs the touch screen input on top of the screen?

We should add that and this code should do the trick 😉 

bool touchInBox(touchPosition touch, int x, int y, int w, int h){
int tx=touch.px;
int ty=touch.py;
u32 kDown = hidKeysDown();
if (kDown &amp; KEY_TOUCH &amp;&amp; tx &gt; x &amp;&amp; tx &lt; x+w &amp;&amp; ty &gt; y &amp;&amp; ty &lt; y+h){
return true;
}else{
return false;
}
}

That or this code lol :/

#include <3ds.h>
#include <stdio.h>
 
int main(int argc, char **argv) {
    gfxInitDefault();
    consoleInit(GFX_TOP, NULL);
 
    // Main loop
    while (aptMainLoop())
    {
        hidScanInput();
       
        touchPosition touch;
        hidTouchRead(&touch);
        printf("\x1b[0;0HX coordinate: %i       ",touch.px);
        printf("\x1b[1;0HY coordinate: %i       ",touch.py);
 
        u32 kHeld = hidKeysHeld();
        if (kHeld & KEY_TOUCH)
            printf("\x1b[2;0HIs Active: TRUE ");
        else
            printf("\x1b[2;0HIs Active: FALSE");
 
        u32 kDown = hidKeysDown();
        if (kDown & KEY_START) break;
 
        gfxFlushBuffers();
        gfxSwapBuffers();
 
        //Wait for VBlank
        gspWaitForVBlank();
    }
 
    gfxExit();
    return 0;
}

6100m, 2018-05-10:
Either should work if sfd2lib is implemented properly. Should compile just fine with devkitpro.

Hikari-chin, 2018-05-10:
sf2dlib is an old deprecated library (though it was pretty bad even when it was new), it definitely won't become a Luma3DS dependency.

6100m, 2018-05-10:
Hell no. I know tons of developers who still use it.

Just going to close this and report abuse you.. I'm so sick of being harrassed along with 102 other developers.

closed 2018-05-10 by 6100m

Hikari-chin, 2018-05-10:
That it's still used doesn't make it any less deprecated, xerpi even said as much in the [sf2dlib README](https://github.com/xerpi/sf2dlib/blob/master/README.md).

6100m, 2018-05-10:
All reported five times.

RIP Hikari-Chin-Pee.

ihaveamac, 2018-05-10:
@6100m stop acting like a child. github issues are not a place for your temper tantrums.