LumaTeam/Luma3DS #1081: 3ds help

Alexhupp, 2018-05-11:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

Dimensional, 2018-05-11:
Whats the matter boy? Timmy fell down the well? Does Flo need change for a dollar (Progressive Insurance Joke)?

Silliness aside, what exactly is the problem? A generic "HELP" message with no details as to exactly what went wrong, only what you have configured and what system, gives nothing for us to help you with.

Alexhupp, 2018-05-11:
What i get is a black screen with only the blue light on if i try and boot into recovery mode still a black screen but wifi indicator turns on i have uninstalled the cfw many times and reinstalled because i was bored but now its on a black screen either its my N2ds hating me or my dsi forwarders i installed killing me

Qyriad, 2018-05-11:
This is not a support forum.

closed 2018-05-11 by Hikari-chin