LumaTeam/Luma3DS #1082: Locale Help

Falcon16, 2018-05-11:
**System model:**

New 3ds

**SysNAND version (+emu/redNAND version if applicable):**

11.5.0-38U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

9.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (1)

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
I recently tried to emulate JPN on my copy of Mario & Luigi: Superstar Saga + Bowser's Minion's and wondering it won't.

**Steps to reproduce:**

1. Follow the guide for locale emulation.

**Dump file:**

MasterGamingYT, 2019-04-06:
You should update luma and your 3ds first

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH