LumaTeam/Luma3DS #1087: New 3ds XL don't boot. Always show a stack dump or another error.

pilladoll, 2018-05-17:
Console: New 3ds XL
Sysnand 11.6.0-39
Entrypoint Bootstrap
Luma 3ds version: was on 8.1.1 but also occurs with 9.0 both stable.
**Luma3DS configuration/options:**
Default EmuNAND: (1 )
Screen brightness: (4 )
Splash: ( )
PIN lock: ( )
New 3DS CPU: ( )
--
Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: (x )
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: ( )

**Explanation of the issue:**

Hi, hope anyone can help us fixing this.

My friend's New 3ds XL was working fine, until a week ago. The console sometimes displayed some messages abot an sd card error (most of the time, when he played games). The system showed the arm11 error and the stack dump was generated.  He just reboots the console and continued playing.

Monday he lends me the console to show me the error, asking for my help. Just checking the error (errtype: Generic) and with the messages about the sd card error I told him he needed to replace the sd card. Using the card of my camera, the system worked fine, and  with no errors for almost 2 hours. But he keeped playing with his defective SD card, until yesterday. When the same arm11 error showed, he took out the sd and turned off the console, put back the sd and the stack dump showed just powering on the console. He can access to luma config, and to the chainloader menu.

We tried cleaning the menu extdata but same error occurs.

If the sd card (a new one, not the damaged) is inserted, the system show this:
https://imgur.com/a/rRIdvQH

But if the sd is not inserted, then the generic stack dump appears on screen.

**Steps to reproduce:**

1. Just power on the console 
2. If there's no sd inserted, the console shows a stack dump. If the sd is inserted show the loader error

thisismike97, 2018-05-18:
> he took out the sd and turned off the console

You're saying he took out the SD card and **afterwards** turned off the system? because that would be prettyyyy badd.

You could try reinstalling Boot9Strap however since in your screenshot it speaks of a loader error.

> He can access to luma config, and to the chainloader menu.

Does this mean you can still access Godmode9? Because if you can you could also try to restore the NAND backup.

(This is not an issue with luma3ds though, so this isnt really the right place to seek help.)

pilladoll, 2018-05-19:
Yes. He pulled out the sd before powering off the console.

And, yes, he can access godmode9, but I don't know if he have a Nand backup

SmackBro, 2018-05-25:
@thisismike97 Dude,he can't just reinstall b9s,it is still on his console,uninstalling it in that state might kill the system entirely,if ntrboot doesn't work there. The only thing he could do is updating b9s,but I doubt that this will help,as b9s and luma do what they are doing,as he can access godmode9 and the luma config,the first error shows menu as process,so perhaps his homemenu is corrupted due to his dying sd card.

@pilladoll I'd suggest you to here: https://3ds.hacks.guide/troubleshooting and do the "blackscreen on sysnand" sections or you friend does this,if this didn't help and he doesn't have a NAND Backup,you could also go here: https://3ds.hacks.guide/region-changing and try to restore the system with a ctrtransfer image of the his console type and region,so he has a n3dsxl,so you need the n3ds + your region ctrtransfer image. It doesn't matter that it is a 11.5 image,just update the system afterwards,this might also get rid of all his data,but I think that's a little loss for recovering his system.

thisismike97, 2018-05-26:
@SmackBro You clearly dont understand how B9S works if you think a re-install would uninstall it.

He can safely chainload SafeB9Sinstaller and reinstall B9S in the consoles current state, this is likely the best choice to try and fix it if he DOESNT have his NAND backup to restore.

@pilladoll First, find out if your buddy has his NAND backup, if he has it, restore it using GM9. However if he doesnt have his nandbackup you can indeed try the troubleshooting section or a re-install of the cfw and B9S, in the worst case you could do a CTR transfer however i dont know much about it.

SmackBro, 2018-05-26:
I do know what b9s is and how it works,you just described a function of the safeb9sinstaller.

From my everyday computer experience "reinstalling" something goes hand in hand with uninstalling the program first and then reinstalling it,from this perspective,you'd totally fuck the 3ds up.

Since you just wrote "You could try reinstalling Boot9Strap however since in your screenshot it speaks of a loader error." I assumed you meant like "run the uninstall cfw scrips and then install b9s again.",so my apologies,for misinterpreting your message. 

You could say a ctrtransfer image is a clean NAND Backup without any console specific files,because of this you can restore it on any consoles,which enabled us to restore another regions image on our consoles (region changing). A ctrtansfer image would be the last resort,if you brick,deleted your NAND backup,or your NAND backup were corrupted.

pilladoll, 2018-06-07:
Thanks for your help, guys.

I think he -somehow- damaged the motherboard. Did two CTRTransfes (one with the correct region, other with another) and the console still show the arm11 error.